Ergodox-io/ErgoDox #4: Version of KiCad to open the PCB files?

daprice, 2018-03-26:
When I try to open the board in pcbnew, I get the following error:

```
Error
Error loading board.
KiCad was unable to open this file, as it was created with a more recent version than the one you are running. To open it, you'll need to upgrade KiCad to a more recent version.

Date of KiCad version required (or newer): 08/15/2016

Full error text:
PARSE_ERROR: Expecting 'clearance, trace_width, via_dia, via_drill, uvia_dia, uvia_drill, or add_net' in input/source
'/Users/daleprice/Development/Keyboards/AlterErgo/ErgoDOX pcb/ErgoDOX.kicad_pcb'
line 185
offset 6
from dsnlexer.cpp : Expecting() : line 369
```

Except I *do* have a new enough version of KiCad to meet that requirement, 4.0.7 (the latest stable release) from 8/28/2017. Is there a weirdly specific older build I need to dig up in order to open this? KiCad doesn’t seem to keep an archive of older versions, nor did they actually have a release on 8/15/2016 as far as I can tell.

marcong95, 2018-04-21:
I met the same issue as yours, but I found where to download older version of KiCad, you can have a look on [http://downloads.kicad-pcb.org/windows/stable/](http://downloads.kicad-pcb.org/windows/stable/).
Although I have found where to download the older version, I have not tried any of those yet, so I cannot tell you which exact version you need to open the project.

robotmaxtron, 2018-06-05:
I can confirm it's working on r13060-d3c41dd2 which I've been using from the kicad copr repo: https://copr.fedorainfracloud.org/coprs/g/kicad/kicad/

closed 2019-05-03 by robotmaxtron