Ergodox-io/ErgoDox #5: Any plans to modify the design for kailh pcb sockets and quick swap switches?

dmgm, 2018-05-12:
Most people who build a keyboard are also interested in trying different switches, and sometimes mixing switches on a board. [Kailh pcb sockets](https://novelkeys.xyz/products/kailh-pcb-sockets) seem to be a popular way to achieve this, used by the input club and ergdox ez. Are their any plans to produce a socket compatible PCB design?

robotmaxtron, 2018-05-15:
I think that's a great idea.

dmgm, 2018-05-17:
Other benefits are in long term maintenance, you can remove all switches to get at the board if something needs to be re-soldered or replaced. 

I've seen a kailh socket footprint file somewhere, but I doubt its as simple as cut and paste. could someone with pcb designe experience comment.

lingster, 2019-08-27:
hi, this looks like an interesting request that I would like to tackle. For the switch replacement is the target item: Kailh_socket_optional_reversible?

evan-chaney, 2020-11-30:
Hi I know this is ancient, but it's something people may still be interested in. I will throw a vote towards Kailh_socket_optional_reversible being the correct one to implement. Or perhaps a hotswap and a non-hotswap version to keep PCB costs down?