Ergodox-io/ErgoDox #10: Left hand dwg files won't open

JSchiltz19, 2019-05-09:
The left-hand dwg files for the normal acrylic case won't open using AutoCAD or dwg true view. I also can't convert them to svg's.

robotmaxtron, 2019-05-11:
Which files specifically? I just pulled them down and was able to open them in FreeCAD.

n8gray, 2019-12-01:
Maybe related: The scale of the left hand dxf file is all wrong.  It shows the board as about 4m wide.

thedragonsmith, 2021-03-06:
I was unable to open several of the left hand files, too when I tried to import into inkscape.  The file opened fine on Autodesk viewer.  When I found this issue here and tried the right plate file, it opened without error in inkscape.  I compared the two and found that the right file used periods where the left file used commas.  I replaced them in the left file and that seems to have solved the problem:  
`155,43284551931851` now looks like `155.43284551931851`
The drawing LOOKS okay now that I've got it open, but I'm not 100% certain I didn't overwrite an important comma

The right files appear to have a lot more in the Header as well but I don't know enough about DXF to know what that's doing.