Ergodox-io/ErgoDox #15: Gerber files only include the right hand side of the keyboard.

mjparme, 2020-12-04:
The gerber files only contain the right hand side of the keyboard. Are there gerber files available for the left hand side? 

For that matter it looks like the KiCad `.pcb` file only has the right hand side as well. Am I missing something? Or maybe I am supposed to mirror it somehow and then generate left hand side gerbers?

I have a feeling this is a dumb question and people are going to laugh when they read it.

jp94260, 2020-12-04:
Hi 
The pcb is with the both side, if you flip it you will see the left side
Regards

mjparme, 2020-12-04:
> Hi
> The pcb is with the both side, if you flip it you will see the left side
> Regards

Ahh ok. I get it now! There is no difference between right and left, you just need two of them and you flip one over.

closed 2020-12-04 by mjparme